cvra/robot-software #1: Kernel Panic: scheduler priority assert

nuft, 2016-09-21:
original [Issue](https://github.com/cvra/master-firmware/issues/36) by @Stapelzeiger:

while running a bootloader ping through the master board it crashed at an assert at chsys.h:318.
The comment above the assert says:

>
 The following condition can be triggered by the use of i-class functions
     in a critical section not followed by a chSchResceduleS(), this means
     that the current thread has a lower priority than the next thread in
     the ready list.
ch.rlist.r_current has prio 3 (must be the TCP thread),
ch.rlist.r_queue.p_next is the uavcan thread and has prio 64

I quickly checked all use of I class functions in our code and didn't spot any obvious problems.
the two thread structures look ok, so no memory corruption of the thread working areas.

antoinealb, 2016-10-26:
@SyrianSpock Did you observe this in your testing using recent versions of the code ?

SyrianSpock, 2016-10-26:
Never happened in my testing (since august).

antoinealb, 2016-10-27:
Ok, then I will close it for now. We can always reopen it later if it shows up again and we have more information.

closed 2016-10-27 by antoinealb