cvra/robot-software #3: Motor board config loading

SyrianSpock, 2016-10-08:
I think the config loading should be documented somewhere in this repo.
Or even better, write a python package with tools to interact with a motor board (like we have for the CAN bootloader).

antoinealb, 2017-03-30:
I have started working on it as a part of my journey to clean UAVCAN. I plan to modify the motor board firmware so that it stores its config in flash, then write a python script to write the config to all boards from the USB dongle. Everything should happen through UAVCAN's parameter service, which will allow us to deprecate our custom mechanism.

SyrianSpock, 2017-08-21:
@antoinealb This is still in progress ?

antoinealb, 2017-08-21:
It was still controversial when we discussed it last time, but I think most of the team was in favor of it. Haven't made progress on the code front, but still happy to do it.

SyrianSpock, 2017-08-21:
I still think it's a good idea to do it.

SyrianSpock, 2017-12-11:
It's getting painful to debug things on the master board as it panics when we reboot it from gdb (so the bus is not rebooted and the motor config setting times out). I think we should implement this soon.

closed 2018-01-20 by SyrianSpock