cvra/robot-software #9: Two different logging subsystems

antoinealb, 2016-10-27:
So we have something in `log.h` and something in `aversive/error.h`. One logging system should be enough for the whole robot I think ;)

I think Aversive's one is more feature rich, we can be both good and bad. In any case it should be cleaned up and maybe simplified (There are definitely some features we don't need).

What do you guys think ? Any reason to keep our own logger ?

I volunteer to cleanup/test/put everywhere Aversive's one should we decide to keep it by the way.

SyrianSpock, 2016-10-27:
I think it would be nice to use the Aversive logging, we may even use it to trigger some behaviors for example panic on an error.

nuft, 2016-10-27:
Ok, why not. I also like the dynamic enabling of the different severity levels.

Some suggestions for if you want to touch the code:
- There are too many error severity levels. I wouldn't know which one to choose. What do you think of only 3 levels like "INFO, WARNING and ERROR"?
- I think error numbers are useless. I prefer human readable messages.
- It would be nice to keep the timestamp and thread name in the logs, as we're doing now.

antoinealb, 2016-10-27:
I think having `DEBUG` separate from `INFO` and turned off by default makes sense, as it will contain for example the state on every iteration of a compute loop or something like that. I agree for removal of `EMERG` though.

> I think error numbers are useless. I prefer human readable messages.

Agree, they will be removed.

> It would be nice to keep the timestamp and thread name in the logs, as we're doing now.

yup, won't touch that.

PR #10 (2016-10-28): Unify logging subsystem
This PR fixes #9 by moving everything to be managed by Aversive's error handler. There are still a few things I would like to do, including runtime selection of log verbosity and calling functions information that I would like to add. Some cleanup could be useful as well. However, I feel like this work is already useful and therefore can be merged.

@SyrianSpock can you test on hw please ?
closed 2016-10-28 by SyrianSpock