cvra/robot-software #13: What are {build,test}.sh ?

antoinealb, 2016-11-06:
I see that all the individual projects are littered with `{build,test}.sh` files, that seem to sort-of automate the build of the project. I feel like this belongs to the developer's own automation and we should not put that kind of stuff in the repository, but I would like to hear the team's own rationale for this: what are your feelings on this guys ?

antoinealb, 2016-11-06:
On the same idea: does anybody actually use `.sublime-project` files ? They are pretty obviously broken in their current state.

nuft, 2016-11-06:
This is just basic laziness for not having to type:
```
packager
make dsdl
make
```
My impression is that everyone is using those scripts.

Also, the idea is to have exactly the same sequence for automated and local builds.

Stapelzeiger, 2016-11-06:
We don't need multiple test scripts since with packager all the tests can be compiled together anyway. So one at the root would be enough.
The build scripts are quite useful though, because to build the firmware correctly multiple steps are usually needed. I created them after having bad builds because I forgot one of the commands.

Stapelzeiger, 2016-11-06:
For the sublime-project file, I am not using it. I suggest we remove it.

antoinealb, 2016-11-06:
ok I removed the sublime files.

For the build files I don't think this makes the build process simpler, it only makes it more obscure IMHO. Especially since you rarely need to run `packager` or `make dsdlc`, at least in my experience.

Also I would prefer a way to ensure that *exactly* the same commands are needed to build the firmware for every subproject to ease onboarding of new team members, and those shell scripts don't go in that direction I think.

nuft, 2016-11-06:
> We don't need multiple test scripts since with packager all the tests can be compiled together anyway. 

I agree.
Though, even if not optimal, the current test setup works.
If someone is motivated to get the packager to build two separate firmwares but only one test binary,  feel free to do so.

SyrianSpock, 2016-11-06:
I don't use these scripts. I have to agree with @antoinealb, using one-liner build commands with `&&` covers my daily use.
The only scripts we need are those in the `ci` directory.

@nuft I wouldn't merge the two boards unittests into one single binary. But having a simple way to build all firmwares/tests for all boards would be nice to have.

nuft, 2016-11-06:
ok for me

antoinealb, 2016-11-07:
For me one reason to avoid using those scripts is that they make it harder to add build time options, e.g. you have no way of doing the equivalent of `make USE_BOOTLOADER=no`, which in turn leads to more complicated READMEs such as:

> Set `USE_BOOTLOADER = no` in the Makefile, then build & lauch openocd:

Where do I put that ? Why do I have to edit the Makefile ? etc. etc.

compare to : `make clean && make USE_BOOTLOADER=no`. Which one is simpler ?

nuft, 2016-11-08:
I removed the files.

closed 2016-11-08 by nuft