cvra/robot-software #21: Move robot out of obstacle (if any) before planning a new path

SyrianSpock, 2016-12-21:
Sometimes the opponent is detected very close to the robot and it will think it's inside the opponent bounding box obstacle. The robot should be able to first get out of this situation before recomputing a path to its goal.

antoinealb, 2017-04-12:
We also need to handle the case where we are in a static obstacle or outside the table bouding box.

SyrianSpock, 2018-04-15:
One way to handle this:
- When blocked in obstacle, set a flag in the robot state
- Provide GOAP with a safe way to get itself out of the blocked state
- Let GOAP do the work of actually getting out when needed

antoinealb, 2018-04-15:
Maybe as a source of inspiration, how microb (authors of aversive) did it in 2009: https://github.com/onitake/aversive/blob/master/projects/microb2009/mainboard/strat_avoid.c#L233