cvra/robot-software #23: Robot specific parameters storage and loading

SyrianSpock, 2016-12-21:
We need to choose how to store and load the configuration of different robots. Do we use `config.yaml` for everything and remove all the defines.
There are common parameters and ones that are specific to each robot. How do we separate those ?

antoinealb, 2016-12-21:
Do you have an example for common parameters ?

I would put everything in config.yaml but keep default values define'd. I think we could separate in 3 files:

* config_common.yml
* config_debra.yml
* config_small_robot.yml

We will always load parameters from the common file. Then we will load the robot specific one.

SyrianSpock, 2016-12-21:
> Do you have an example for common parameters ?

All calibrated parameters (wheelbase, encoder pulses per mm, etc)

antoinealb, 2016-12-21:
Aren't those ones the perfect examples of per-robot parameters ?

SyrianSpock, 2016-12-21:
Completely misread the question :|
Common parameters are all non-calibrated parameters (motor wheel radius, motor track length), map / strategy specific stuff (opponent obstacle size)

antoinealb, 2016-12-21:
ok makes more sense !

SyrianSpock, 2017-01-23:
For now, we'll keep one config file for each robot.

closed 2017-01-23 by SyrianSpock