cvra/robot-software #52: Unit tests fail on Clang

antoinealb, 2017-05-16:
For some reason Travis version of clang has a broken c++ library. I disabled Clang's build in ae47b7c but it would be good to re enable it later.

Possibly relevant links:

* https://github.com/philsquared/Catch/issues/334
* https://github.com/Andersbakken/rtags/issues/204

antoinealb, 2018-04-04:
Builds on my laptop with Clang trunk but some tests fail. Would be really nice to have it working in CI.

antoinealb, 2018-04-04:
We could even enable [AdressSanitizer](https://github.com/google/sanitizers/wiki/AddressSanitizer) to hunt for memory being used before init. I think something similar exists for GCC (`-fsanitize=address`).

SyrianSpock, 2018-04-04:
I would be keen on activating all the sanitizers https://clang.llvm.org/docs/SanitizerSpecialCaseList.html

SyrianSpock, 2018-04-29:
Fixed by #133

closed 2018-04-29 by SyrianSpock