cvra/robot-software #53: Build fails on ARM GCC 6.3.1

SyrianSpock, 2017-08-21:
http://178.32.216.117:8000/cvra/robot-software/294

SyrianSpock, 2017-08-21:
Updated ChibiOS, doesn't seem to fix it.
[This thread](http://www.chibios.com/forum/viewtopic.php?t=1839) suggests it's related to the `exit` function being called somewhere.

antoinealb, 2017-08-21:
I am a bit surprised that this is triggered by an update. Did newlib change at the same time or something ?

nuft, 2017-09-11:
in [newlib fini.c](https://github.com/bminor/newlib/blob/e0f24404b3fcfa2c332ae14c3934546c91be3f42/newlib/libc/misc/fini.c#L20)  the function `_fini()` is declared as follows:
```
extern void _fini (void);
```
It is called in `__libc_fini_array()` after the cleanup routines.

Just defining `_fini()` gets rid of the liker issue. I didn't find out what it's actually supposed to do. Though in case of master-firmware we will never call `exit()` so it shouldn't be a problem.

What's strange is that this problem appears only now. Where was `_fini()` defined before?

antoinealb, 2017-09-13:
> What's strange is that this error appears only now. Where was _fini() defined before?

`_fini` was not called before:

```
$  arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ ../tools/call_explorer.py build/ch.list _fini
trajectory_finished  -> trajectory_angle_finished
trajectory_has_ended -> trajectory_finished
```

@SyrianSpock could you take a look at who's calling `_fini` using `call_explorer.py` please ?

antoinealb, 2017-09-13:
I think this might be due to exceptions, because `exit` is called by `abort`.

SyrianSpock, 2017-09-13:
As @nuft mentioned `_fini` is called by `__libc_fini_array` which is called after the end of `main()`. [Relevant stackoverflow thread](https://stackoverflow.com/questions/15265295/understanding-the-libc-init-array).
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ ../tools/call_explorer.py build/ch.list _fini
trajectory_finished  -> trajectory_angle_finished
trajectory_has_ended -> trajectory_finished
__libc_fini_array    -> _fini
```

antoinealb, 2017-09-13:
Grrr, I don't understand what changed and why `_fini`. Meh, I guess we can just add _fini by hand (#55).

SyrianSpock, 2017-09-14:
I really don't understand, the proximity beacon firmware seems to build fine without defining `_fini()` ....

antoinealb, 2017-09-14:
motor control firmware is also fine apparently.

antoinealb, 2017-09-14:
Wait no, I just checked after picking gcc 6.3 and neither motor nor proximity are building. Are they for you ?

SyrianSpock, 2017-09-14:
Motor control firmware builds if I define _fini, see #55

SyrianSpock, 2017-09-14:
So if I define _fini in the main of motor control firmware, proximity beacon firmware builds fine (otherwise it fails too)...

commit d4ffb273 (2017-09-14): Merge pull request #55 from cvra/fix-build-gcc-6-3-1

Fix build for GCC 6.3
Resolves #53
closed 2017-09-14 by SyrianSpock