cvra/robot-software #100: Move config defines to header file

nuft, 2018-03-17:
Many configuration parameters for different subsystems are accumulating in the Makefile.
In my opinion this is a bad practice and those parameters should be independent of the build system. 
I propose to move them into header files grouped by topic.

example from master-firmware:
```make
# List all default C defines here, like -DDEBUG
DDEFS += -DUAVCAN_TOSTRING=0 \
		 -DUAVCAN_STM32_NUM_IFACES=1 \
		 -DUAVCAN_STM32_TIMER_NUMBER=2 \
		 -DUAVCAN_STM32_CHIBIOS=1 \
		 -DUAVCAN_TINY=0 \
		 -DLWIP_DEBUG=1 \
		 -DSHELL_MAX_ARGUMENTS=5 \
		 -DSHELL_CMD_THREADS_ENABLED=FALSE \
		 -DSHELL_CMD_TEST_ENABLED=FALSE
````

antoinealb, 2018-03-17:
I am not sure if you can change them other than by putting them in the build system though. For example where would you put `#define UAVCAN_STM32_NUM_IFACES 1` ? It is internal to uavcan and cannot easily be injected.

pavel-kirienko, 2018-03-17:
GCC has a flag named `-include` which is very useful exactly for this purpose: https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html