cvra/robot-software #105: The USB stack of ChibiOS may crash the firmware under certain circumstances

pavel-kirienko, 2018-03-22:
I hope you can tolerate my intrusion here, I'm just looking for friendly open source projects that leverage the ChibiOS's USB stack. The reason is that there appears to be a serious issue with the USB stack that may crash the OS under certain conditions; it is described here: http://www.chibios.com/forum/viewtopic.php?f=25&t=4568

So far the issue was observed on ChibiOS 16.1 running on STM32F105, STM32F373, and STM32F446. As can be seen in the linked thread, Giovanni is reluctant to take action until the issue is confirmed on newer revisions of the RTOS. Seeing as you're using version 17.6, perhaps you could spare half an hour to run a simple test described in the thread? That would be supremely useful for ChibiOS users in general.

Thanks!

SyrianSpock, 2018-03-22:
Hi Pavel,
Sure we can try to reproduce the issue. However this week is busy for us. We'll have a look on Monday.

pavel-kirienko, 2018-03-27:
The issue has been confirmed by Giovanni. Look out for updates!

closed 2018-03-27 by pavel-kirienko

SyrianSpock, 2018-03-28:
Good to know, thanks :+1: