cvra/robot-software #122: Make logging great again

SyrianSpock, 2018-04-15:
We have a lack of logging on our robots:
- We need to log more, use `DEBUG` for internal logging.
- We need to enable SD card storage of logs again.
- We need to be able to retrieve saved logs via ethernet (reenable the http server).

antoinealb, 2018-04-15:
I think this is really high priority and should be done for Eurobot. There we won't have a lot of testing opportunities, and being able to trace through the execution of the code is really important.

I think it is really needed that every action & decision of the robot has at least one NOTICE message that we can trace on the LCD screen. `DEBUG` is really for inner loops, e.g. in the A* graph search algorithm.

Currently the LCD screen is enabled after color selection, making it more complicated to see messages soon. We should just init it in main() before strategy.

SyrianSpock, 2018-04-15:
Agreed. This is something we need to fix in this week and use for our testing and tuning for Eurobot.

antoinealb, 2018-04-19:
Just realized that an alternative to HTTP would be... UAVCAN: https://github.com/UAVCAN/dsdl/tree/master/uavcan/protocol/file

I don't think it is too hard to implement correctly.

nuft, 2018-04-19:
Yes, why not. I think it's a good Idea to use UAVCAN as the main protocol to interface with our boards.
This avoids reimplementing the same functionality over and over again.

nuft, 2018-04-19:
> - We need to enable SD card storage of logs again.

If I remember correctly, the SD card isn't working on the Nucleo144 shield or at least needs some debugging.

antoinealb, 2018-04-19:
Another question remains: how do we map a match to a log file ? We do not have a realtime clock to provide a time reference.

Just name them 001.txt, 002.txt, and so on ? And display the filename on the LCD so that you can take note of it when starting up the robots? Not very convenient though.

SyrianSpock, 2018-04-19:
Can't you assume the latest log corresponds to the latest game? All valid games should have a starting log message announcing the start of the game, and ending log, so we could parse for that in the tool we use to fetch log files. We can try something dumb, and improve it later on as we encounter pain points.

antoinealb, 2018-04-19:
So one option would be to start logging to SD card at the beginning of the game. I assumed you would start a new log file at every boot, meaning there would be log files that do not correspond to a game.

SyrianSpock, 2018-04-19:
Sounds good

closed 2018-10-25 by SyrianSpock