cvra/robot-software #146: Setup on-board computer using SaltStack

SyrianSpock, 2018-08-12:
We need to configure the on-board computer (Intel NUC) for the rover.
Packages required:
- [] Python with data analysis packages
- [] ROS with perception packages
- [] Realsense specific packages
- [] CAN-USB ready
- [] WiFi setup for remote access

antoinealb, 2018-08-12:
Some thoughts:

* Use Ubuntu server edition to avoid all the crap that comes with desktop edition?
* Use hostapd to have the robot create a hotspot instead of the phone tethering stuff we did before?
* SocketCAN / SLCAN support is pretty easy to have with systemd: https://gist.github.com/sebastianhaas/f134398df1c710fcb8eaf297f49d33d4
* I would suggest having some cantools and tshark if we need packet dump abilities.
* I would also suggest some kind of monitoring platform such as InfluxDB/Telegraf. This can help with debugging system issues as well as monitoring CPU load, temp, etc. I have a salt state for that if you need it.

antoinealb, 2018-09-20:
Questions:

* What are the needed python packages ?
* What are the ROS perception packages ?

SyrianSpock, 2018-09-20:
For python I would go for python 3 with numpy/scipy (ping @froj what do you need here?) and CVRA stack (bootloader, yaml, serial)

For ROS, I would go for `ros-melodic-perception` metapackage and then install the realsense SDK see https://github.com/SyrianSpock/machine/blob/master/salt/roots/dev/realsense.sls you will need to disable the GUI when building the SDK and not install GTK packages.

PR #155 (2018-09-23): Rover NUC setup
This PR implements the NUC setup for the rover (closes #146).

Things I could not test:

* Realsense camera driver
* SocketCAN instantiation (I know that one is not completely finished, but it should work with just a change in the systemd service file).

@SyrianSpock I will give you the PC tomorrow. If you bring a CAN dongle I will be able to fix the SocketCAN thingie before handing it over.