cvra/robot-software #185: Setup a wireless debug server

SyrianSpock, 2019-02-14:
To make testing on the table less painful, it would be nice to setup a wireless debug server (with a Raspberry Pi) to which we would wire:
- the STLink for firmware upload
- the CAN USB adapter to monitor CAN traffic
- the USB shell

antoinealb, 2019-02-15:
some suggestions:

* power the raspi through the CAN bus
* use openocd GPIO adapter instead of st link, might work better?

Lu-ni, 2019-04-19:
- [x] openocd GPIO for firmware upload
 - [x] the CAN USB adapter to monitor CAN traffic
 - [x] the USB shell
 - [ ] Bridge ethernet traffic on wlan (to have udp packet remotly)
 - [ ] Print a case and make the hardware cleaner

antoinealb, 2019-10-22:
was useful to teach, wont finish as we got stuff to do.

closed 2019-10-22 by antoinealb