cvra/robot-software #191: UWB beacon firmware is built with exceptions enabled

nuft, 2019-03-08:
I noticed that for the uwb beacon all the exception handling code is linked with the firmware.

We should investigate and try to remove the cause as it blows up the binary size and makes debugging harder.

antoinealb, 2019-03-08:
Could it be enough to add c++ stubs ? https://github.com/cvra/robot-software/blob/master/can-io-firmware/src/stubs.cpp

nuft, 2019-03-08:
no, because Eigen calls `operator new()` which results in an error in `stubs.cpp`. 

I'm still investigating where the exceptions are coming from...

antoinealb, 2019-03-08:
I think if you have `operator new()`, you will have exceptions brought in, as `new` might throw... Best to understand why Eigen is pulling dynamic memory allocation.

nuft, 2019-03-08:
Apparently, it depends on the object size.
In the following example Eigen starts using malloc/new for n >= 8. (compiled for ARM)

```
    Eigen::Matrix<float, n, n> A;
    Eigen::Matrix<float, n, n> ATA = A.transpose() * A;
```

nuft, 2019-03-08:
I'm surprised since I believed that basic operations in Eigen don't use malloc.
Turns out that's wrong 😕

antoinealb, 2019-03-08:
Disappointing... But I think there are some ways to prevent Eigen from using malloc()

antoinealb, 2019-03-08:
Apparently its possible to do it using `Eigen::Map` instead. Pinging resident Eigen expert @SyrianSpock 

https://stackoverflow.com/questions/21402426/c-eigen-matrix-operations-vs-memory-allocation-performance

SyrianSpock, 2019-03-09:
If we start doing memory allocation ourselves to avoid Eigen using malloc, the code is gonna get very ugly. Eigen is using malloc for temporaries, one way to reduce them is to use the `noalias()` method (http://eigen.tuxfamily.org/dox/TopicWritingEfficientProductExpression.html). 

I prefer to study the problem more, using `EIGEN_NO_MALLOC` to trigger assert failures on malloc calls, we can use that to understand what's happening. One way to tweak allocation would be implementing our own allocator to replace `aligned_malloc` implementation in Eigen (https://github.com/cvra/eigen/blob/9e97af7de76716c99abdbfd4a4acb182ef098808/Eigen/src/Core/util/Memory.h#L150-L171)

nuft, 2019-03-09:
I checked agian, actually `operator new()` is not called. (I think because our matrix dimensions are small enough)

However, I found that overwriting `std::__throw_bad_function_call()` removes the exception handling code.

PR #192 (2019-03-12): Unify C++ stubs
Moved C++ stubs to lib/chibios-syscalls to address #57. I observed that for the UWB firmware the bin is 40kB smaller (236556 to 194972 bytes). May fix #191
closed 2019-03-12 by SyrianSpock