cvra/robot-software #194: UWB panics at decawave initialization after reset.

nuft, 2019-03-14:
Some UWB beacons hang when resetting/power-cycling.
The error comes from a failed `dwt_initialise()` in this [line](https://github.com/cvra/robot-software/blob/master/uwb-beacon-firmware/src/decawave_interface.c#L98).

It keeps failing when further power-cycling.
My hypothesis is that the decawave radio keeps some state during power cycling due to the large decoupling capacitors (Quickly shorting those seems to properly reset the chip and it works again).

nuft, 2019-03-14:
@antoinealb proposed to add a reset cycle before initialization the using `UWB_RST_N` line.

antoinealb, 2019-03-14:
Try using PC9 for reset ? I hope it should fix the issue. We did not use it because of cvra/uwb-beacon-board#2

nuft, 2019-03-14:
Yes, it works. Fixed in 6171fe128d31fc6b8dbc8408a50a69a890b46eb3.

closed 2019-03-14 by nuft