cvra/robot-software #226: Robust communication between robots

SyrianSpock, 2019-05-14:
Before #189 we have observed in 2019 that UWB was not reliable enough to communicate the position between robots. So our robots ended up colliding into each other which made our game strategy fail. We have a few ideas related to power supply and heating issues that need investigation. By the end of this we should have a robust real-time communication of position between our robots.

antoinealb, 2019-05-14:
For me this is actually two parts:

1. We need boards that do not overheat and maintain a "reasonable" service. That means no overheating, no dieing when the bus voltage is a bit low, etc.
2. We need a high level protocol to cope with packet loss. Since we now have a working IP over UWB implementation, we might use TCP for this directly (which is nice), or build something more complicated à la raft.

SyrianSpock, 2019-05-14:
Agreed, I think UWB boards' design could use a thorough review, and we can experiment with converting supply voltages on the board itself (using the +12V supply)

antoinealb, 2019-05-14:
Yes, I wanted to do a review using the newly found knowledge about heating. My guess would be somewhere on the power supply path something is bad.