cvra/robot-software #228: Review the core electronics of the robots

SyrianSpock, 2019-05-14:
We've observed some flakiness with some core elements in the electronics of our robots:
- [ ] the control panel is more fragile and needs a real PCB to be designed
- [ ] the screen seems flaky, although most issues disappeared after resetting from the master board. We need to investigate its shock tolerance and sensitivity to power supply noise
- [ ] the RAM on the master board is getting scarcer, we should investigate STM32F7 boards that would allow more RAM!

antoinealb, 2019-06-02:
wrt point 3 (RAM getting lower), we can work on two other aspects:

1) Not over-provision thread stacks
2) Use "better" algorithms, unfortunately GOAP is quite memory hungry, maybe we can do better somehow.

nuft, 2019-06-03:
don't forget, F7 has a more complicated memory architecture, which brings issues with DMA.
I would wait updating.

SyrianSpock, 2019-06-03:
I only mentioned "investigate STM32F7", don't freak out ;)

SyrianSpock, 2019-06-03:
> Not over-provision thread stacks

sure needs some cleanup

> Use "better" algorithms, unfortunately GOAP is quite memory hungry, maybe we can do better somehow.

To be investigated too, this is an open question

antoinealb, 2020-06-25:
Closing this as we are migrating away from the master board, so part 3 (low ram) is not an issue any more.

part 1 (control panel pcb) is done, and lives at https://github.com/cvra/front-panel-board

closed 2020-06-25 by antoinealb